hku-mars/r2live #1: great!but how can I upgrade my mid-40😢,I have an external imu.

fkdmmm, 2021-02-25:
It's really great, but can you please tell me how to integrate mid40 with external imu, because I have done it according to your algorithm a month ago, according to the original loam version (although your algorithm has been upgraded many times ) I want to upgrade now. It is too extravagant for me to buy a lidar with imu built-in again. I have a nine-axis imu. How can I upgrade? Thanks, the poor from the northeast of the mainland, thank you again

ziv-lin, 2021-02-28:
For Livox Mid-40, you need to carefully calibrate both the extrinsic and temporal offset between LiDAR and IMU. With this, you can use the external IMU for compensating the motion distortion of point cloud data.

fkdmmm, 2021-03-01:
> For Livox Mid-40, you need to carefully calibrate both the extrinsic and temporal offset between LiDAR and IMU. With this, you can use the external IMU for compensating the motion distortion of point cloud data.

Doctor, sincere thanks from the heart; but can you tell me in detail how to pass the existing LIVOX algorithm program (your organization: the MARS laboratory of the University of Hong Kong, or LIVOX official SDK), because I have not completed the programming and algorithm personally Modification ability, just imitate and simply change the parameters of these tasks (if you can tell me about the process in detail, I would appreciate you more) Which algorithm is the most suitable for me to upgrade MID-40 to add an external IMU? Many thanks. I am just a surveying and mapping worker, and I am very grateful for the help of high knowledge and technology through you, and the sincere gratitude from another world.

fkdmmm, 2021-03-01:
[http://img1.gtimg.com/news/pics/hv1/120/249/2148/139737315.jpg](url)

fkdmmm, 2021-03-01:
[image]

closed 2021-07-14 by stale[bot]