hku-mars/r2live #3: typo in equ(16) ?

narutojxl, 2021-03-30:
Hi authors,
Thanks for your hard-core paper and sharing your knowledge very very much! Is there a typo in equ(16) ? Thanks for your help and time very much!
[image]
[image]

ziv-lin, 2021-03-30:
Yes,  we have found this error in our paper. We make some mistakes in
"copying" formulation in the codes to our paper. We will correct this
mistake in the next version of the paper...

narutojxl, 2021-03-30:
OK, thanks author, my problem is solved, close it now.

closed 2021-03-30 by narutojxl