hku-mars/r2live #4: Fδx and Fw derivation

narutojxl, 2021-03-31:
Hi author,
I'm curious about Fδx and Fw derivation in Supplementary Material B. There are some question want to ask for your help, hope this will not make you annoyed and angry. Thanks for your help and time a lot!
* in the following picture, the delta_* variables in time i instant, is with hat ? 
[image]

* When deriving rotaion part, this is my derivation, compared with paper's result, there are some differences: 
 1. right jacobian does not transpose and is multiplied with delta_t.
 2. accoring to noise variable w_i defination, Fw(0, 0) and Fw(0, 2)  seems should exchange position.
[image]
[image]
[image]

* When deriving translation part, there seems exist `-0.5*delta_t*delta_t` in Fδx(1, 6) and Fw(1,1).
* The velocity result mach with paper's result, this is the derivation.
[image]
[image]

ziv-lin, 2021-03-31:
Let me check them again, we also found some other errors in our paper and we will correct them in the next version of this paper.
Anyway, thank you for your correction.

narutojxl, 2021-03-31:
Thanks author, Indeed, the above all equations depends on the correct equ(6) of paper, we need very carefully derive it. I will follow author's idea to see whether i can get the same result with paper's result of equ(6). Thanks again!
[image]

ziv-lin, 2021-03-31:
Yes, you are right, we need to careful in deriving such formulations. Actually speaking, we started to write this paper one week before DDL. It is unavoidable for us in making mistakes (e.g. typos and errors in derivations) in this version of preprint paper. 
I will upload an amendation version of this paper this week to this GitHub page.
Again, thank you for your corrections~

closed 2021-04-05 by narutojxl