hku-mars/r2live #6: Requirements (Software + Hardware).

Camilochiang, 2021-04-11:
Hey guys! Great work.

Could you please already indicate already your requirements? (at least which LiDAR, Ubuntu and ROS version)
It will be helpfully to prepare the material in the meanwhile..

Thanks!

fangchuan, 2021-04-20:
I have same question of the hardware configuration like camera . The camera seems work well when walking from indoor to outdoor scene. Could you tell me what kind of camera you are using in the paper? @ziv-lin @XW-HKU @Ecstasy-EC 
Thanks in advance!

ziv-lin, 2021-04-20:
Sorry for the late reply, the software requirement of our experiments is:
1. Ubuntu > 16.04 
2. ROS
3. Ceres-solver 1.4
4. PCL>=1.9
  For the hardware settings, we assume that the camera, LiDAR, and IMU sensor are rigidly connected, and the extrinsics between LiDAR and IMU are well-calibrated (we use the LiDAR built-in IMU, the extrinsics are known).  
The camera we use is the FLIR RGB camera (Blackfly S BFS-u3-13y3c), and is set the mode as auto exposure when we sampling the data.

closed 2021-04-20 by Camilochiang

mbcel, 2021-04-24:
And how did you synchronize the Livox Lidar with the camera? Since no GPS device is used (right?) the Livox ros driver does not give an option for synchronzing timestamps (only supports GPS)

I could think of somehow creating a running mean of the difference between the Livox clock and ROS clock (host clock). For example by comparing the receive ros time and livox package timestamp and averaging this. However, this will then have an average constant offset of the travel time between Livox and host which would experimentally have to be estimated (or maybe as another varibale in the Kalman filter?) 

Or did you write a patch for the Livox ros driver to also support ppt synchronization?

duanqg, 2021-05-27:
livox support ptp synchronization

seajayshore, 2021-06-30:
You say you use "Blackfly S BFS-u3-13y3c" as the camera but this is a USB 3.0 camera and does not support PTP time synchronisation. 
Can you say how you achieve time synchronisation with the camera?
Or do you actually use the GigE Vision version of the camera?

ziv-lin, 2021-06-30:
We online estimated the time delay therefore the time synchronization is not necessary. However, you need to make sure that all data are sample in the same time base (e.g. all of our bags files are time stamped with our ubuntu OS time).

antun008, 2021-09-28:
Can someone exactly explain how can all bag files be stamped with ubuntu os time?

Camilochiang, 2021-09-28:
hei @antun008 .
1. Is a good practice to start a new issue if you need something.
2. Is a good practice to comment in a open issue. This issue has been closed.
3.  You could either use the LIVOX ros driver version of the same author that has been modified for this or do it your self a posteriori. Read point 5 of the README.md  ....