hku-mars/r2live #12: Could not find #include <vins_estimator/States.h>

wwtinwhu, 2021-06-21:
and #include <vins_estimator/Pose6D.h>

ziv-lin, 2021-06-21:
It seems that I have forgotten to add some files that used to generate ros_message header files, let me fix it now.

wwtinwhu, 2021-06-21:
> It seems that I have forgotten to add some files that used to generate ros_message header files, let me fix it now.

yeah，thx！

ziv-lin, 2021-06-21:
I have found the problem and commit a hotfix just now, can you try it again?

wwtinwhu, 2021-06-21:
> I have found the problem and commit a hotfix just now, can you try it again?

So fast！I will try it again tomorrow！Good night.

wwtinwhu, 2021-06-22:
> I have found the problem and commit a hotfix just now, can you try it again?

It works! Good Work!
[image]

closed 2021-06-22 by wwtinwhu