hku-mars/r2live #13: issue

Alexdev585, 2021-06-21:
ERROR: cannot launch node of type [feature_tracker/feature_tracker]: Cannot locate node of type [feature_tracker] in package [feature_tracker]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [r2live/lio_feat_extract]: Cannot locate node of type [lio_feat_extract] in package [r2live]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [r2live/r2live]: Cannot locate node of type [r2live] in package [r2live]. Make sure file exists in package path and permission is set to executable (chmod +x)

How do I resolve this? thanks

ziv-lin, 2021-06-21:
It seems that you have not sourced the path, try:
source ~/catkin_ws/devel/setup.bash

Alexdev585, 2021-06-21:
yes i have entered this but still the same result
LOAM-Livox package has the same error now aswell?

can you assist?

Camilochiang, 2021-06-21:
Your problem is that you are sourcing different packages from HKU in the same environment. I recommend you to check the order how you are doing it or avoid it if is possible ( I had a similar issue before).

Alexdev585, 2021-06-21:
how would I avoid doing that?
thanks for your help, by the way

Camilochiang, 2021-06-21:
Please reboot your machine or refresh the environment and paste the code that you are executing (line by line)

Alexdev585, 2021-06-21:
cd catkin_ws
catkin_make
source ~/catkin_ws/devel/setup.bash

roslaunch r2live demo.launch

result
Base path: /home/alex/catkin_ws
Source space: /home/alex/catkin_ws/src
Build space: /home/alex/catkin_ws/build
Devel space: /home/alex/catkin_ws/devel
Install space: /home/alex/catkin_ws/install
WARNING: Package name "GrabImage" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
####
#### Running command: "cmake /home/alex/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/alex/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/alex/catkin_ws/install -G Unix Makefiles" in "/home/alex/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/alex/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/alex/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.23
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
WARNING: Package name "GrabImage" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 10 packages in topological order:
-- ~~  - GrabImage
-- ~~  - camera_model
-- ~~  - dnb_msgs
-- ~~  - feature_tracker
-- ~~  - r2live
-- ~~  - camera_control_msgs
-- ~~  - pylon_camera
-- ~~  - point_cloud_io
-- ~~  - rviz_camera_stream
-- ~~  - usb_cam
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'GrabImage'
-- ==> add_subdirectory(grabimage)
Traceback (most recent call last):
  File "/opt/ros/melodic/share/catkin/cmake/order_paths.py", line 33, in <module>
    main()
  File "/opt/ros/melodic/share/catkin/cmake/order_paths.py", line 28, in main
    with open(args.outfile, 'w') as fh:
IOError: [Errno 13] Permission denied: '/home/alex/catkin_ws/build/grabimage/catkin_generated/ordered_paths.cmake'
CMake Error at /opt/ros/melodic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/home/alex/catkin_ws/build/catkin_generated/env_cached.sh
  "/usr/bin/python2" "/opt/ros/melodic/share/catkin/cmake/order_paths.py"
  "/home/alex/catkin_ws/build/grabimage/catkin_generated/ordered_paths.cmake"
  "--paths-to-order" "/opt/ros/melodic/include" "/usr/include"
  "/opt/ros/melodic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp"
  "--prefixes" "/home/alex/catkin_ws/devel" "/home/alex/catkin_ws/src"
  "/opt/ros/melodic") returned error code 1
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/list_insert_in_workspace_order.cmake:29 (safe_execute_process)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:108 (list_insert_in_workspace_order)
  grabimage/CMakeLists.txt:10 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/alex/catkin_ws/build/CMakeFiles/CMakeOutput.log".
[... 62 lines cut ...]
 * /r2live/config_file: /home/alex/catkin...
 * /r2live/vins_folder: /home/alex/catkin...
 * /record_bag_name: /home/ziv/kitti_b...
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
    feature_tracker (feature_tracker/feature_tracker)
    lio_feat_extract (r2live/lio_feat_extract)
    r2live (r2live/r2live)
    rvizvisualisation (rviz/rviz)

auto-starting new master
process[master]: started with pid [3792]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to cd6c5c60-d2d5-11eb-a926-00d861a8852d
WARNING: Package name "GrabImage" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
process[rosout-1]: started with pid [3803]
started core service [/rosout]
ERROR: cannot launch node of type [feature_tracker/feature_tracker]: Cannot locate node of type [feature_tracker] in package [feature_tracker]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [r2live/lio_feat_extract]: Cannot locate node of type [lio_feat_extract] in package [r2live]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [r2live/r2live]: Cannot locate node of type [r2live] in package [r2live]. Make sure file exists in package path and permission is set to executable (chmod +x)
process[rvizvisualisation-5]: started with pid [3806]

Camilochiang, 2021-06-21:
> yes i have entered this but still the same result
> LOAM-Livox package has the same error now aswell?
> 
> can you assist?

What about this ? Did you restart your environment? 

Is either that or you have some permission issues as say in your code...

Alexdev585, 2021-06-21:
> > yes i have entered this but still the same result
> > LOAM-Livox package has the same error now aswell?
> > can you assist?
> 
> What about this ? Did you restart your environment?
> 
> Is either that or you have some permission issues as say in your code...

yes its the same, I restarted my environment 

I cant seem to resolve the permission errors, any ideas? anyone

Camilochiang, 2021-06-22:
Just out of curiosity, Where do you have Livox ros driver and Livox SDK?

ziv-lin, 2021-06-30:
To install Livox-ros-driver and SDK, please follow this guide:
https://github.com/Livox-SDK/livox_ros_driver

closed 2021-06-30 by ziv-lin