hku-mars/r2live #19: camera shutter and dark environment

duanqg, 2021-06-27:
It's really a great job!
 I have two questions about the camera here. 
Question 1: Do the shutter shutter and the global shutter of the camera (the camera FLIR Blackfly you used is the global shutter) have a great influence on the experimental results? (Theoretically speaking, the shutter at low speed will not have too much deformation deformation will not be too large). Question 2: does the dark environment have a great impact on the construction of the map, and if so, how can it be set to reduce the impact? thank you

duanqg, 2021-06-27:
Another question is whether there is an inheritance relationship between R2Live and Fastlio, because I have done GPS optimization based on Fastlio, and I wonder if it is feasible to do some experiments on the basis of R2Live

ziv-lin, 2021-06-30:
Notice that I have replied to your email about these questions before. By this, I decide to close this issue~.

closed 2021-06-30 by ziv-lin