hku-mars/r2live #20: use with velodyne 16 issue

sihanh, 2021-06-29:
when I use Velodyne 16, switch lidar type to 2, the feature extract node has the following warning.

> Failed to find match for field 'normal_x'.
> Failed to find match for field 'normal_y'.
> Failed to find match for field 'normal_z'.
> Failed to find match for field 'curvature'.

I don't think the velodyne_points field has those attributes and I cannot find in the source code of this warning.

ziv-lin, 2021-07-01:
Eh, I have not tested my algorithm on Velodyne-16 yet and therefore I do not promise it can work well with it. Maybe you can try to modify the code of LiDAR feature extraction and LIO subsystem to make it capable work with Velodyne LiDARs.

closed 2021-07-01 by ziv-lin

satyajitghana, 2021-07-16:
@ziv-lin 

Hi, i can see that FAST_LIO supports VELO16, below is the code, where `normal_x,` `normal_y`, `normal_z` and curvature values are there.

https://github.com/hku-mars/FAST_LIO/blob/669eedfc7df1f64413b84adaef96e54b6747e169/src/preprocess.cpp#L317-L326

So do i have to just upgrade the FAST_LIO version in this repository to make it work?

CanCanZeng, 2021-07-21:
Hi @satyajitghana  @sihanh , do you eventually get it work with vlp16?