hku-mars/r2live #23: Is it compatible with opencv 4?

aditdoshi333, 2021-06-30:
Thank you for sharing your amazing work. I am trying to catkin_make r2live but it is failing and I guess the error is related to opecv version. Currently I am using opencv 4.4.

ziv-lin, 2021-06-30:
To work with OpenCV 4.X, you need to modified some header (about the includes) and some C++ API interface of the VIO subsystem. 
Please refer to the link provides as follow:
https://blog.csdn.net/tjj1998/article/details/110500320
https://wenda.ncnynl.com/article/5

aditdoshi333, 2021-06-30:
Hey, thanks for the reply. I will try the steps for sure. Can you please share your Opecv and Ceres versions?

Is there a DOCKERFILE of the project?

ziv-lin, 2021-07-01:
My openCV Version is: 3.3.1
My ceres version is: 1.14
My Eigen version is the newest one: 3.3.9

closed 2021-07-16 by stale[bot]