hku-mars/r2live #25: Launch error

aditdoshi333, 2021-07-01:
Hello,

I am able to build the project with opecv4 changes it is successful. But when I running

roslaunch r2live demo.launch

It is crashing and giving the following error:

Multi thread started 
/home/adit/catkin_ws/src/r2live/r2live/src/./fast_lio/fast_lio.hpp 791
[Ros_parameter]: fast_lio/dense_map_enable ==> 1
[Ros_parameter]: fast_lio/lidar_time_delay ==> 0
[Ros_parameter]: fast_lio/max_iteration ==> 4
[Ros_parameter]: fast_lio/fov_degree ==> 360
[Ros_parameter]: fast_lio/filter_size_corner ==> 0.4
[Ros_parameter]: fast_lio/filter_size_surf ==> 0.4
[Ros_parameter]: fast_lio/filter_size_surf_z ==> 0.4
[Ros_parameter]: fast_lio/filter_size_map ==> 0.4
[Ros_parameter]: fast_lio/cube_side_length ==> 1e+09
[Ros_parameter]: fast_lio/maximum_pt_kdtree_dis ==> 0.5
[Ros_parameter]: fast_lio/maximum_res_dis ==> 0.3
[Ros_parameter]: fast_lio/planar_check_dis ==> 0.1
[Ros_parameter]: fast_lio/long_rang_pt_dis ==> 50
[Ros_parameter]: fast_lio/publish_feature_map ==> 0
/home/adit/catkin_ws/src/r2live/r2live/src/./fast_lio/fast_lio.hpp 816
/home/adit/catkin_ws/src/r2live/r2live/src/./fast_lio/fast_lio.hpp 829
/home/adit/catkin_ws/src/r2live/r2live/src/./fast_lio/fast_lio.hpp 831
[ WARN] [1625127750.923615584]: waiting for image and imu...
~~~~/home/adit/catkin_ws/src/r2live/r2live/ doesn't exist
[r2live-4] process has died [pid 33098, exit code -11, cmd /home/adit/catkin_ws/devel/lib/r2live/r2live __name:=r2live __log:=/home/adit/.ros/log/7a2ab4f8-da45-11eb-a931-3df647905bdb/r2live-4.log].
log file: /home/adit/.ros/log/7a2ab4f8-da45-11eb-a931-3df647905bdb/r2live-4*.log

aditdoshi333, 2021-07-01:
It is throwing a file doesn't exist error. But I am not able to locate the respective files. 

https://github.com/hku-mars/r2live/blob/cb6d2cc8f62c28f45a189b9053e35f71070c665b/r2live/src/fast_lio/fast_lio.hpp#L875
https://github.com/hku-mars/r2live/blob/cb6d2cc8f62c28f45a189b9053e35f71070c665b/r2live/src/fast_lio/fast_lio.hpp#L875

aditdoshi333, 2021-07-01:
[feature_tracker-2-stdout.log](https://github.com/hku-mars/r2live/files/6747033/feature_tracker-2-stdout.log)
[lio_feat_extract-3-stdout.log](https://github.com/hku-mars/r2live/files/6747036/lio_feat_extract-3-stdout.log)
[master.log](https://github.com/hku-mars/r2live/files/6747037/master.log)
[roslaunch-imperfection-33055.log](https://github.com/hku-mars/r2live/files/6747038/roslaunch-imperfection-33055.log)
[rosout.log](https://github.com/hku-mars/r2live/files/6747039/rosout.log)
[rosout-1-stdout.log](https://github.com/hku-mars/r2live/files/6747040/rosout-1-stdout.log)
[rvizvisualisation-5-stdout.log](https://github.com/hku-mars/r2live/files/6747041/rvizvisualisation-5-stdout.log)

These are the log files.

ziv-lin, 2021-07-01:
I have seen that somebody else has successfully run my codes on their own computer, for examples:
https://blog.csdn.net/learning_tortosie/article/details/118144287
https://github.com/hku-mars/r2live/issues/12

So I believe it is most probably caused by the differences in your software environment. By this I suggest:
1. Update your Eigen version to the newest one first.
2. Download ceres-solver, check to branch 1.14.X, build and install the ceres-solver.
3. Recomple our source codes and try again.

My software version are listed as follow:
My OpenCV Version is: 3.3.1
My ceres version is: 1.14
My Eigen version is the newest one: 3.3.9

aditdoshi333, 2021-07-01:
Thank you for replying. I wanted to ask one more thing if I change the topic to /livox_pcl of the point cloud in rviz. I am able to visualize the map and it looks similar to yours. But the issue is with the memory consumption within few seconds it consumes around 10+ gb of ram. 

[image]

ziv-lin, 2021-07-01:
The point cloud data with the topic name "livox_pcl" is the **raw point cloud** sampled from that LiDAR sensor, is has not been registered in the maps. You should display the point cloud after registered.

Watson-OU-221, 2021-07-07:
Hello! Could I ask you how to deal with the problem that you mentioned? I met the same problem. Is it the reason that the OpenCV 4.X is incompatible with them?

> Hello,
>
> I am able to build the project with opecv4 changes it is successful. But when I running
>
> roslaunch r2live demo.launch
>
> It is crashing and giving the following error:
>
> Multi thread started [...]

rahulsharma11, 2021-07-08:
yes, with opencv4, built successfully but not able to get /**cloud_registered** data. Only **livox_pcl0** data is available.
Downgrading to opencv3, able to get the map with **/cloud_registered.**

Not sure the main reason.

One more point is with realsense lidar l515, not able to get **/cloud_registered** data at all with the compatible opencv version also but still able to get **raw_point_cloud** data.

zgxsin, 2021-07-13:
> Hello! Could I ask you how to deal with the problem that you mentioned? I met the same problem. Is it the reason that the OpenCV 4.X is incompatible with them?
> 
> > Hello,
> > I am able to build the project with opecv4 changes it is successful. But when I running
> > roslaunch r2live demo.launch
> > It is crashing and giving the following error:
> > Multi thread started
> > /home/adit/catkin_ws/src/r2live/r2live/src/./fast_lio/fast_lio.hpp 791
> > [Ros_parameter]: fast_lio/dense_map_enable ==> 1
> > [Ros_parameter]: fast_lio/lidar_time_delay ==> 0
> > [Ros_parameter]: fast_lio/max_iteration ==> 4
> > [Ros_parameter]: fast_lio/fov_degree ==> 360
> > [Ros_parameter]: fast_lio/filter_size_corner ==> 0.4
> > [Ros_parameter]: fast_lio/filter_size_surf ==> 0.4
> > [Ros_parameter]: fast_lio/filter_size_surf_z ==> 0.4
> > [Ros_parameter]: fast_lio/filter_size_map ==> 0.4
> > [Ros_parameter]: fast_lio/cube_side_length ==> 1e+09
> > [Ros_parameter]: fast_lio/maximum_pt_kdtree_dis ==> 0.5
> > [Ros_parameter]: fast_lio/maximum_res_dis ==> 0.3
> > [Ros_parameter]: fast_lio/planar_check_dis ==> 0.1
> > [Ros_parameter]: fast_lio/long_rang_pt_dis ==> 50
> > [Ros_parameter]: fast_lio/publish_feature_map ==> 0
> > /home/adit/catkin_ws/src/r2live/r2live/src/./fast_lio/fast_lio.hpp 816
> > /home/adit/catkin_ws/src/r2live/r2live/src/./fast_lio/fast_lio.hpp 829
> > /home/adit/catkin_ws/src/r2live/r2live/src/./fast_lio/fast_lio.hpp 831
> > [ WARN] [1625127750.923615584]: waiting for image and imu...
> > ```
> > 
> > [r2live-4] process has died [pid 33098, exit code -11, cmd /home/adit/catkin_ws/devel/lib/r2live/r2live __name:=r2live __log:=/home/adit/.ros/log/7a2ab4f8-da45-11eb-a931-3df647905bdb/r2live-4.log].
> > log file: /home/adit/.ros/log/7a2ab4f8-da45-11eb-a931-3df647905bdb/r2live-4*.log
> > ```

I have the same issue with opencv 4.x

GuoPingPan, 2021-07-22:
Hello, I used my livox_horizon radar and realsenseD435 camera to reproduce your code. I did not calibrate imu and set estimate_extrinsic: 2 in filr_cam.yaml. The following problems occurred:
setting /run_id to bd46673e-ead8-11eb-951f-00e93a01d3ad
process[rosout-1]: started with pid [23836]
started core service [/rosout]
process[feature_tracker-2]: started with pid [23843]
process[lio_feat_extract-3]: started with pid [23844]
process[r2live-4]: started with pid [23845]
process[rvizvisualisation-5]: started with pid [23851]
[INFO] [1626950218.102325585]: init begins
[INFO] [1626950218.105416167]: Loaded config_file: /home/nvidia/r2live/src/r2live/r2live/../config/filr_cam.yaml
result path /vins_result_no_loop.csv
 not exists, trying to create it
Failed to create folder
[INFO] [1626950218.106288960]: ROW: 480.000000 COL: 640.000000
[WARN] [1626950218.106301497]: have no prior about extrinsic param, calibrate extrinsic param
[INFO] [1626950218.106328359]: Unsynchronized sensors, online estimate time offset, initial td: 0
[r2live-4] process has died [pid 23845, exit code -11, cmd /home/nvidia/r2live/devel/lib/r2live/r2live __name:=r2live __log:=/home/nvidia/.ros/log/bd46673e -ead8-11eb-951f-00e93a01d3ad/r2live-4.log].
log file: /home/nvidia/.ros/log/bd46673e-ead8-11eb-951f-00e93a01d3ad/r2live-4*.log

How can I solve it?

Vox09, 2021-07-28:
Hello, I just find out the problem could be solved by setting the optimization level in `CMakeLists.txt` to be 0, but the performance will be a bit slower.
`set(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} -std=c++17 -O0 -w")`
Looks like the compiler optimization did something bad.
I bet it disturbed some synchronization operations.

BTW, my env is Ubuntu 20.04 with ROS-noetic and OpenCV 4.x

Vox09, 2021-07-28:
Although the program could run by setting `-O0`, it acts very weirdly...

A proven better solution is to install g++-7. `sudo apt install g++-7`.
And then set the compiler in the `CMakeLists.txt`
```
set(CMAKE_C_COMPILER "/usr/bin/gcc-7")
set(CMAKE_CXX_COMPILER "/usr/bin/g++-7")
```
So that you can keep the optimization level to be 3. And the algorithm seems to work properly.

closed 2021-08-12 by stale[bot]

rainlord, 2021-11-25:
> Hello,
>
> I am able to build the project with opecv4 changes it is successful. But when I running
>
> roslaunch r2live demo.launch
>
> It is crashing and giving the following error:
>
> Multi thread started [...]

can you tell me how do you fix this problem?

xuwuzhou, 2022-01-06:
> Although the program could run by setting `-O0`, it acts very weirdly...
>
> A proven better solution is to install g++-7. `sudo apt install g++-7`. And then set the compiler in the `CMakeLists.txt` [...]

Good Job, Bro!

Hjw-Edward, 2022-06-13:
Hello! Could I ask you how to deal with the problem that you mentioned? I met the same problem. It is worth mentioning that I verify my software environment.
My software version are listed as follow:
My OpenCV Version is: 3.3.1
My ceres version is: 1.14.0
My Eigen version is: 3.3.9

when I roslaunch r2live demo.launch,it gives  the following error:
... logging to /home/hjw/.ros/log/179a6104-eae2-11ec-9b09-f02f74f50fd8/roslaunch-hjw-pc-19758.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hjw-pc:43993/

SUMMARY
========

PARAMETERS
 * /acc_mul_G: 0
 * /fast_lio/cube_side_length: 1000000000
 * /fast_lio/dense_map_enable: True
 * /fast_lio/filter_size_corner: 0.4
 * /fast_lio/filter_size_map: 0.4
 * /fast_lio/filter_size_surf: 0.4
 * /fast_lio/filter_size_surf_z: 0.4
 * /fast_lio/fov_degree: 360.0
 * /fast_lio/local_loop_win: [110, -116, 155, ...
 * /fast_lio/m_long_rang_pt_dis: 500
 * /fast_lio/map_file_path:  
 * /fast_lio/max_iteration: 4
 * /fast_lio/maximum_pt_kdtree_dis: 0.5
 * /fast_lio/maximum_res_dis: 0.3
 * /fast_lio/planar_check_dis: 0.1
 * /fast_lio/publish_feature_map: False
 * /feature_extraction/N_SCANS: 6
 * /feature_extraction/blind: 0.1
 * /feature_extraction/cos160: 160.0
 * /feature_extraction/disA: 0.01
 * /feature_extraction/disB: 0.1
 * /feature_extraction/edgea: 2
 * /feature_extraction/edgeb: 0.1
 * /feature_extraction/group_size: 8
 * /feature_extraction/inf_bound: 4
 * /feature_extraction/jump_down_limit: 8.0
 * /feature_extraction/jump_up_limit: 170.0
 * /feature_extraction/lidar_type: 1
 * /feature_extraction/limit_maxmid: 6.25
 * /feature_extraction/limit_maxmin: 3.24
 * /feature_extraction/limit_midmin: 6.25
 * /feature_extraction/p2l_ratio: 225
 * /feature_extraction/point_filter_num: 1
 * /feature_extraction/point_step: 3
 * /feature_extraction/smallp_intersect: 172.5
 * /feature_extraction/smallp_ratio: 1.2
 * /feature_extraction/using_raw_point: 1
 * /feature_tracker/config_file: /home/hjw/catkin_...
 * /feature_tracker/image_topic: /camera/image_color
 * /feature_tracker/vins_folder: /home/hjw/catkin_...
 * /if_dump_log: 0
 * /if_lidar_start_first: 1
 * /if_write_to_bag: 0
 * /imu_topic: /livox/imu
 * /lidar_drag_cam_tim: 10
 * /r2live/config_file: /home/hjw/catkin_...
 * /r2live/vins_folder: /home/hjw/catkin_...
 * /record_bag_name: /home/ziv/kitti_b...
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    feature_tracker (feature_tracker/feature_tracker)
    lio_feat_extract (r2live/lio_feat_extract)
    r2live (r2live/r2live)
    rvizvisualisation (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[feature_tracker-1]: started with pid [19788]
process[lio_feat_extract-2]: started with pid [19789]
process[r2live-3]: started with pid [19790]
process[rvizvisualisation-4]: started with pid [19796]
[ INFO] [1655132785.463453570]: init begins
[ INFO] [1655132785.466717995]: Loaded config_file: /home/hjw/catkin_ws/src/r2live/r2live/../config/filr_cam.yaml
result path /vins_result_no_loop.csv
 not exists, trying to create it 
Failed to create folder  
[ INFO] [1655132785.467430591]: ROW: 512.000000 COL: 640.000000 
[ WARN] [1655132785.467442893]:  fix extrinsic param 
[ INFO] [1655132785.467660198]: Extrinsic_R : 
  0.00193065   -0.0166655     0.999859
   -0.999998  0.000312504   0.00193612
-0.000344726    -0.999861   -0.0166649
[ INFO] [1655132785.467680142]: Extrinsic_T : 
 0.03056  0.06429 -0.00865
[ INFO] [1655132785.467691981]: Unsynchronized sensors, online estimate time offset, initial td: 0
[Ros_parameter]: /lidar_drag_cam_tim ==> 10
[Ros_parameter]: /acc_mul_G ==> 0
[Ros_parameter]: /if_lidar_start_first ==> 1
[Ros_parameter]: /if_write_to_bag ==> 0
[Ros_parameter]: /if_dump_log ==> 0
[Ros_parameter]: /record_bag_name ==> /home/ziv/kitti_bag/lic_record_bag/lic.bag
Multi thread started 
/home/hjw/catkin_ws/src/r2live/r2live/src/./fast_lio/fast_lio.hpp 786
[Ros_parameter]: fast_lio/dense_map_enable ==> 1
[Ros_parameter]: fast_lio/lidar_time_delay ==> 0
[Ros_parameter]: fast_lio/max_iteration ==> 4
[Ros_parameter]: fast_lio/fov_degree ==> 360
[Ros_parameter]: fast_lio/filter_size_corner ==> 0.4
[Ros_parameter]: fast_lio/filter_size_surf ==> 0.4
[Ros_parameter]: fast_lio/filter_size_surf_z ==> 0.4
[Ros_parameter]: fast_lio/filter_size_map ==> 0.4
[Ros_parameter]: fast_lio/cube_side_length ==> 1e+09
[Ros_parameter]: fast_lio/maximum_pt_kdtree_dis ==> 0.5
[Ros_parameter]: fast_lio/maximum_res_dis ==> 0.3
[Ros_parameter]: fast_lio/planar_check_dis ==> 0.1
[Ros_parameter]: fast_lio/long_rang_pt_dis ==> 50
[Ros_parameter]: fast_lio/publish_feature_map ==> 0
/home/hjw/catkin_ws/src/r2live/r2live/src/./fast_lio/fast_lio.hpp 811
/home/hjw/catkin_ws/src/r2live/r2live/src/./fast_lio/fast_lio.hpp 824
/home/hjw/catkin_ws/src/r2live/r2live/src/./fast_lio/fast_lio.hpp 826
[ WARN] [1655132785.507232763]: waiting for image and imu...
~~~~/home/hjw/catkin_ws/src/r2live/r2live/ doesn't exist

[r2live-3] process has died [pid 19790, exit code -11, cmd /home/hjw/catkin_ws/devel/lib/r2live/r2live __name:=r2live __log:=/home/hjw/.ros/log/179a6104-eae2-11ec-9b09-f02f74f50fd8/r2live-3.log].
log file: /home/hjw/.ros/log/179a6104-eae2-11ec-9b09-f02f74f50fd8/r2live-3*.log