hku-mars/r2live #35: Questions about dynamic scenes

zxd123, 2021-08-02:
I don't see anything in this paper about dynamic scene processing, but in the demonstration to see a lot of dynamic objects in the scene can also get better results, want to ask why?

ziv-lin, 2021-08-03:
Firstly, since we have the prior estimation in each LIO update or VIO update, the outliers caused by the dynamic objects play a limited effect on the state estimation.
Secondly, for the VIO sub-system (base on VINS-MONO), we use the RANSAC algorithm for rejecting the error feature matching.
Finally, for LiDAR only system, we can also use the outlier rejection introduced in my previous work LOAM-Livox (see https://github.com/hku-mars/loam_livox) to remove the points of moving objects.

zxd123, 2021-08-04:
> Firstly, since we have the prior estimation in each LIO update or VIO update, the outliers caused by the dynamic objects play a limited effect on the state estimation.
> Secondly, for the VIO [...]

Thank you very much for your reply! I will study this work ASAP.

closed 2021-08-19 by stale[bot]