hku-mars/r2live #36: What is the unit of acceleration in the IMU topic?

nerius59, 2021-08-03:
What is the unit of linear acceleration in the IMU topic? 

In livox _ros _driver, the unit of acceleration in the IMU topic is [g] by default. (https://github.com/Livox-SDK/livox_ros_driver/issues/63)
But in your sample bag file(harbor.bag), the unit of acceleration in your IMU topic(/livox/imu) looks like [m/s^2] for me.

Best regards,

ziv-lin, 2021-08-03:
I have forgotten one of the most important things is that I have modified the livox_ros_driver for data sampling. 
Firstly, I have multiplied the amplitude of gravity when publishing the IMU topic. 
Secondly, I modified the timestamp of LiDAR topic from LiDAR time base to  system time base.
I will soon opensource my modified livox_ros_driver and add some instructions in my readme.

ziv-lin, 2021-08-03:
I have released my modified livox_ros_driver just now, can you have a try this one to see if it OK? If everything is OK, I will modify the readme of this repo.
https://github.com/ziv-lin/livox_ros_driver_for_R2LIVE
[image]

aditdoshi333, 2021-08-04:
Hey @ziv-lin ,

I tried the modified livox_ros_driver but I am not able to find the package even after soucing setup.bash and setup.sh

rahulsharma11, 2021-08-04:
Hi @ziv-lin , The new driver you shared does not contains CMakeList.txt.
I used older CMakeList.txt and built successfully. Also able to see the livox_ros_driver node.
Just need to check with raw data of livox.

You can confirm if that is the solution.

ziv-lin, 2021-08-04:
Thank you for your reply and I have found the problem, the CMakeList.txt is ignored by the git when I added them to my repo. I will fix this problem and later add some instructions to the project page of R2LIVE.

closed 2021-08-19 by stale[bot]