hku-mars/r2live #41: Error when I run the demo.launch

xiaobinbin813, 2021-09-05:
When I run demo.launch and the bag file you provided, I encountered the following problems:

> /home/wangbin/catkin_ws/src/r2live/r2live/src/fast_lio/IMU_Processing.cpp, 400, dt = 0.104579
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
[rvizvisualisation-5] process has died [pid 2346, exit code -11, cmd /opt/ros/melodic/lib/rviz/rviz -d /home/wangbin/catkin_ws/src/r2live/r2live/../config/rviz/vins_rviz_config.rviz __name:=rvizvisualisation __log:=/home/wangbin/.ros/log/bfc04080-0e2c-11ec-9a78-000c29ba4d3c/rvizvisualisation-5.log].
log file: /home/wangbin/.ros/log/bfc04080-0e2c-11ec-9a78-000c29ba4d3c/rvizvisualisation-5*.log

And the rviz crashed. What should I do to solve this problem?

closed 2021-09-20 by stale[bot]

Lollipopkkk, 2022-01-17:
do you solve this problem?I meet it,too

mumu2822, 2022-04-20:
the same problem

cuboya, 2024-11-25:
I meet it,too

cuboya, 2024-11-25:
> do you solve this problem?I meet it,too你能解决这个问题吗？我也遇到了

hello，i meet the same problem

cuboya, 2024-11-25:
> the same problem 同样的问题

hello，i meet the same problem