hku-mars/r2live #47: bad map result

huangsiyuan0717, 2021-09-13:
Thanks for your great job!
I use your data, but don't get your result, what can i do to get better result?

[image]

ziv-lin, 2021-09-13:
What is your hardware platform (cpu and RAM) and software environment ( opencv, pcl, eigen, and etc )?

huangsiyuan0717, 2021-09-14:
> What is your hardware platform (cpu and RAM) and software environment ( opencv, pcl, eigen, and etc )?
Thanks for your reply, my hardware and software environment as the follows:
cpu: Intel® Core™ i5-10400 
RAM: 16G
eigen: 3.3.4
opencv: 3.4.0
pcl: 1.8.1

huangsiyuan0717, 2021-09-14:
> What is your hardware platform (cpu and RAM) and software environment ( opencv, pcl, eigen, and etc )?

This issue was closed due to my wrong operation, I open it again.

ziv-lin, 2021-09-14:
It seems that your environment is OK, is every time you always get the wrong results? i.e. you run the same example five times and the map is drifted each time.

huangsiyuan0717, 2021-09-15:
> It seems that your environment is OK, is every time you always get the wrong results? i.e. you run the same example five times and the map is drifted each time.

Yes，I got the same result after three times.
And there are some error as the follows:
[image]
[image]

ziv-lin, 2021-09-15:
Is the other 2 bags run normally?

ziv-lin, 2021-09-16:
Notice that your RAM is only 16Gb, and you are running this bag with rViz displaying all the points, which will take up too much memory. 
Can you use the system monitor to monitor the use of memory when you're running this bag? 
Type this command on your terminal to open the monitor:
```
gnome-system-monitor
```

ziv-lin, 2021-09-16:
Maybe your problem is caused by loss of data due to the limited available memory.

closed 2021-10-01 by stale[bot]