hku-mars/r2live #48: R2Live + external IMU

Camilochiang, 2021-09-13:
Good day!
I have a running environment with FAST-LIO2 using an external IMU, thanks for that!
 Would be possible in the future to use an external IMU also with r2live? will be great for MID40 and MID70.

Thanks.

ziv-lin, 2021-09-14:
I am sorry that I can not give your such an answer since I haven't try to use an external IMU yet.

Camilochiang, 2021-09-14:
Easy. I could send you a rosbag if you would like ;)

zgxsin, 2021-09-14:
I am trying to run R2live with an external IMU. I will update here. 

1. Could you confirm the the IMU and LIDAR frame have identity rotation among them in the livox device you have used?

2. And the translation is hard-coded here: https://github.com/hku-mars/r2live/blob/master/r2live/src/fast_lio/include/common_lib.h#L58?

3. If my IMU has a rotation to my LIDAR frame, i have to change it here: https://github.com/hku-mars/r2live/blob/master/r2live/src/fast_lio/fast_lio.hpp#L208:

  from 
  ```
  Eigen::Vector3d p_global(g_lio_state.rot_end * (p_body + Lidar_offset_to_IMU) + g_lio_state.pos_end);
  ```
  to
  ```
  Eigen::Vector3d p_global(g_lio_state.rot_end * (R_Lidar_to_IMU*p_body + Lidar_offset_to_IMU) + g_lio_state.pos_end);
  ```

closed 2021-09-29 by stale[bot]

Camilochiang, 2021-09-29:
Hei @zgxsin Did you succed? Just curious.

zuowei1998, 2022-01-03:
Hello, @zgxsin Did you succed? my device is robosense16 and mpu6050

Sunny-cpp, 2022-01-20:
> 

I also have the mid-70 and want to run R2Live by using it.
could you share your hardware scheme and rosbag?
thank you!

Sunny-cpp, 2022-01-20:
> Easy. I could share you a rosbag if you would like ;) (IMU + MID70 + Camera)

I also have the mid-70 and want to run R2Live by using it.
could you share your hardware scheme and rosbag?
thank you!