hku-mars/r2live #50: Velodyne Sesnor Input

zgxsin, 2021-09-14:
I have a dataset with velodyne point cloud, which is in the format of `sensor_msgs/PointCloud2`. There exists a velo16_handler:
 https://github.com/hku-mars/r2live/blob/master/r2live/src/fast_lio/feature_extract.cpp#L115. So i changed the lidar_type from 1 to 2: https://github.com/hku-mars/r2live/blob/master/config/lio_config.yaml#L2. Afterwards, i launched the program and replayed my dataset.

I got the following issue:

```
Failed to find match for field 'normal_x'.
Failed to find match for field 'normal_y'.
Failed to find match for field 'normal_z'.
Failed to find match for field 'curvature'.
Failed to find match for field 'normal_x'.
Failed to find match for field 'normal_y'.
Failed to find match for field 'normal_z'.
Failed to find match for field 'curvature'.
```

Does that mean that i have to pre-compute the `normal_x`, `normal_y`, `normal_z` and `curvature` for the velodyne point cloud?
Is there any method in your code to do this?

closed 2021-09-29 by stale[bot]

llxClover, 2021-12-16:
> I have a dataset with velodyne point cloud, which is in the format of `sensor_msgs/PointCloud2`. There exists a velo16_handler: [...]

Have you resolved the question?

farhad-dalirani, 2023-01-18:
It is my question too.