hku-mars/r2live #53: Hardware schema

antun008, 2021-09-22:
Does anyone have a schematic of how to pair all the hardware together? It especially bothers me how to connect the camera and make sync with others ...

borongyuan, 2021-10-05:
The simplest way is using a GigE 2.0 camera, which support IEEE1588 (i.e. PTP synchronization). So all the devices and clocks can be synchronized using [linuxptp](http://linuxptp.sourceforge.net). The camera can be driven using [rc_genicam_camera](https://github.com/roboception/rc_genicam_camera). We also build a handheld device with simplified line design.
[image]
[image]

borongyuan, 2021-10-09:
You can use any brand of camera, as long as you can drive it and manage to achieve hardware synchronization. I use GigE 2.0 camera because one important component of the GigE Vision standards in version 2.0 is the Precision Time Protocol (PTP). PTP is also supported by Livox. So this is the simplest way for hardware synchronization. You don't need to modify the driver or add external trigger hardware. As you can see from the photo, there is only one network cable connected to the camera. It powers the camera, transmits data, and synchronizes the clock.
The link you pasted is a GigE 1.2 camera, which does not support PTP. As far as I know you also can’t synchronize hardware via USB. But almost all types of industrial cameras support external trigger input. So you can connect your hardware like [this](https://github.com/ISEE-Technology/CamVox#32-hard-synchronization), if you are not using a GigE 2.0 camera.
GigE 2.0 cameras are not very popular yet. I'm using a Hikrobot camera, which is left over from another project.
https://www.hikrobotics.com/en/machinevision/productdetail?id=5086&pageNumber=1&pageSize=50
It has relative high fame rate (125.2 fps) and low resolution (720x540). I don’t think current algorithms can make use of such a high frame rate. I would probably choose the following type if I need to build a new one. M12 lenses are also easier to get.
https://www.hikrobotics.com/cn/machinevision/productdetail?id=7456&pageNumber=1&pageSize=20
I got pretty good test results with this set of equipment. This is not r2live but also based on FAST-LIO.

https://user-images.githubusercontent.com/22918715/136655706-7da9c022-2837-4097-bcce-4137e65fdb13.mp4

ToutDonner, 2022-02-20:
> The simplest way is using a GigE 2.0 camera, which support IEEE1588 (i.e. PTP synchronization). So all the devices and clocks can be synchronized using [linuxptp](http://linuxptp.sourceforge.net). [...]

你好，请问你那块精灵四电池板是直接插在电池上然后通过vcc和gnd就可以进行输出的吗

borongyuan, 2022-02-20:
是的

thisparticle, 2024-01-07:
> The simplest way is using a GigE 2.0 camera, which support IEEE1588 (i.e. PTP synchronization). So all the devices and clocks can be synchronized using [linuxptp](http://linuxptp.sourceforge.net). [...]

你好，想请教一下海康威视rc_genicam_camera应该怎么配置ptp同步呀？还有个问题就是ptp是只同步了时钟源，雷达和相机是没有同一时刻采集吗？

chenxiaocongAI, 2024-05-27:
@borongyuan 海康威视的相机如果想设置为slave,是否买你那一款不用任何设置，其默认就是slave模式啊？如果需要进行配置，可以告诉我怎么设置吗

borongyuan, 2024-05-28:
> @borongyuan 海康威视的相机如果想设置为slave,是否买你那一款不用任何设置，其默认就是slave模式啊？如果需要进行配置，可以告诉我怎么设置吗

在海康MVS客户端里有IEEE 1588设置。主机上的ptp配置可参考linuxptp的文档。

thegame3761, 2026-01-15:
@borongyuan can you please tell me the name of this board ? How do you setup this board with your setup? Thank you so much.

[image]