hku-mars/r2live #54: Experiencing large drift while mapping with livox avia

aditdoshi333, 2021-09-27:
Hello,

Thanks for sharing such amazing work. I am experiencing large drift when I am running demo.launch on live stream. I have changed the calibration params in config file. 

For livox, I am using the following driver (https://github.com/ziv-lin/livox_ros_driver_for_R2LIVE)

The following is the set of errors I am gettting:
`/opt/catkin_ws/src/r2live/src/fast_lio/IMU_Processing.cpp, 400, dt = 0.105697
/opt/catkin_ws/src/r2live/src/fast_lio/IMU_Processing.cpp, 135, check_state fail !!!!  5.55778 -9.90081 -10.1357
/opt/catkin_ws/src/r2live/src/fast_lio/IMU_Processing.cpp, 135, check_state fail !!!!  5.55778 -9.90081 -10.1357
/opt/catkin_ws/src/r2live/src/fast_lio/IMU_Processing.cpp, 135, check_state fail !!!!  5.55778 -9.90081 -10.1357
/opt/catkin_ws/src/r2live/src/fast_lio/IMU_Processing.cpp, 135, check_state fail !!!!  5.55778 -9.90081 -10.1357
/opt/catkin_ws/src/r2live/src/fast_lio/IMU_Processing.cpp, 135, check_state fail !!!!  5.55778 -9.90081 -10.1357
/opt/catkin_ws/src/r2live/src/fast_lio/IMU_Processing.cpp, 135, check_state fail !!!!  5.55778 -9.90081 -10.1357
/opt/catkin_ws/src/r2live/src/fast_lio/IMU_Processing.cpp, 135, check_state fail !!!!  5.55778 -9.90081 -10.1357
/opt/catkin_ws/src/r2live/src/fast_lio/IMU_Processing.cpp, 135, check_state fail !!!!  5.55778 -9.90081 -10.1357
/opt/catkin_ws/src/r2live/src/fast_lio/IMU_Processing.cpp, 135, check_state fail !!!!  5.55778 -9.90081 -10.1357
/opt/catkin_ws/src/r2live/src/fast_lio/IMU_Processing.cpp, 135, check_state fail !!!!  5.55778 -9.90081 -10.1357
state_inout |[673.69852] | (-12.992, -47.860, 54.054) | (18.039, -1.217, -9.484) | (5.558, -9.901, 0.000) | (-0.150, -1.614, 0.984) | (-1.413, 3.676, -6.752) 
state_in |[673.69852] | (-12.992, -47.860, 54.054) | (18.039, -1.217, -9.484) | (5.558, -9.901, -10.136) | (-0.150, -1.614, 0.984) | (-1.413, 3.676, -6.752) 
/opt/catkin_ws/src/r2live/src/fast_lio/IMU_Processing.cpp, 135, check_state fail !!!!  5.31588 -10.1107 -0.17032
`

**Everytime after the following error I am getting large drift.**
`[ INFO] [1632737692.276809736]:  big translation
[ WARN] [1632737692.277026159]: failure detection!
[ WARN] [1632737692.281113827]: system reboot!
`

Camilochiang, 2021-10-11:
Hei @aditdoshi333 . I read in R3live that you manage to run R2live. Could you explain me how did you do the Camera to IMU calibration? Just curious as Im having large drifts from the start of my bag (#40 )
I just may mention that this have happens when setting 
`<param` name="if_lidar_start_first" type="int" value="1" /> `
in the launch file. I have still no tried in the other way, but just curious if you have encounter similar problems ? :) Thanks for your help!

Camilochiang, 2021-10-13:
Thanks @aditdoshi333 ! I did using [kalibr](https://github.com/ethz-asl/kalibr). I just may mention that the large drift in my case was due a wrong unit in the acceleration (g instead of ms-2)

Camilochiang, 2021-10-14:
Hei @aditdoshi333 
Question. The link that you post does the calibration between the camera and the LIDAR but they use the calibration of the camera and the IMU. Isn't that a problem ? In my case kalibr does not support the required distortion model (Brown-Conrady Model) so now I'm not sure which tool use...

Greetings

cly20-cqu, 2022-01-17:
Have you solved the problem yet?