hku-mars/r2live #56: LiDAR incoming frame too old, need to be drop!!!

Camilochiang, 2021-10-05:
Hei.
Today your code is using for MID40
` ros::Time ct(ros::Time::now());`
on line 163 of `feature_extract.cpp 
This make it imposible to re run pre recorded bags as the following error show up for every lidar message
`LiDAR incoming frame too old, need to be drop!!!`
`Failed to find match for field 'normal_x'.`
`Failed to find match for field 'normal_y'.`
`Failed to find match for field 'normal_z'.`
`Failed to find match for field 'curvature'.`

Is there any possibility or could you help me to implement a `mid_handler` that can recive CustomMsg so the timestamp is synchronized with the camera and IMU?

Thanks!

Camilochiang, 2021-10-06:
I partially solve the problem setting lidar_type: 1 and N_SCANS to 1 in lio_config.yaml
but the frames of the LiDAR are still trashed.  I did install your special version of Livox_ros_driver and it was used for recording the rosbag. As you may see the time are extremily diferent.

IMU at  : 1633417831.613297939
LIDAR at: 221.298467640
IMU at  : 1633417831.618217945
IMU at  : 1633417831.634017944
IMU at  : 1633417831.635231018
IMAGE at: 1633417831.228661775

In my case IMAGE and IMU are taking the timestamp directly from ros::Time.now()
Do you modify the time of the LIDAR in any other place that the driver? If is taking the ros time of the host, why the values are so low? Thanks in advance!

Tomato1107, 2022-01-14:
> In my case IMAGE and IMU are taking the timestamp directly from ros::Time.now()
> Do you modify the time of the LIDAR in any other place that the driver? If is taking the ros time of the host, why the values are so low? Thanks in advance!

May I ask what is your setup for the camera and external imu? How to sync the external imu and camera with the modified livox_ros_driver?

Tomato1107, 2022-01-14:
@Camilochiang Thanks!