hku-mars/r2live #59: how to align pointcloud map to real picture

huangsiyuan0717, 2021-11-11:
[image]
Hi! How did you align the map? And the real map how do you get(google earth?)? 
I want to display the real data use this way with pointcloud map, but I don't konw how to align it .

ziv-lin, 2021-11-11:
I use photoshop to align the maps, with just scaling the point cloud image.

closed 2021-11-24 by huangsiyuan0717

ccq0905, 2022-07-04:
> [image] Hi! How did you align the map? And the real map how do you get(google earth?)? I want to display the real data use this way with pointcloud map, but I don't konw how to align it .

How do you get pointcloud map, could you share with me?