hku-mars/r2live #63: lio_feat_extract died when using Livox Mid-70

cnhanyuhang, 2021-12-17:
Dear developers, thanks for your amazing work on this multi sensor algorithm.

I'm currently trying to run R2live on my devices, they are Livox Mid-70, Zed2i(camera and built-in IMU). I've customized the filr_cam.yaml to adapt my device, and rename the topics to the same provided in you rosbag. When play my own recorded rosbag. The camera part and imu data have been successfully connected.
[image]

but my lidar data can not be linked.
[image]

It shows this error in terminal:
[image]

I check the log file but it's a blank file. So far I have no idea how to connect my lidar(mid-70) data. Could you help me with this problem? Thanks in advance!

Camilochiang, 2021-12-17:
1. This repo is not even in use as the authors are working in the new version 
2. You dont even see the lidar in the node graph, therefore I will asume that is not been detected/you have bigger issue not related to r2live. Did you install livox ros driver as indicated in the instructions?

cnhanyuhang, 2021-12-17:
Thank you for prompt reply. I think it may not be the driver issue because I have recorded /livox/lidar topic in my rosbag and it can be used in loam-livox. Let me recheck my bag and do more tests, then I may update later. Anyway, thank you.

Best regards,
Han Yuhang

-----------------------
Han Yuhang
Master of Science
School of Mechanical Engineering,
National University of Singapore （NUS）

***@***.***
(65)80387345

cnhanyuhang, 2021-12-17:
> 1. This repo is not even in use as the authors are working in the new version
> 2. You dont even see the lidar in the node graph, therefore I will asume that is not been detected/you have bigger issue not related to r2live. Did you install livox ros driver as indicated in the instructions?

Thank you for prompt reply. I think it may not be the driver issue because I have recorded /livox/lidar topic in my rosbag and it can be used in loam-livox. Let me recheck my bag and do more tests, then I may update later. Anyway, thank you.

ziv-lin, 2021-12-17:
If the ros message can be used in loam-livox, that means you are recording the pointcloud data in PCL/pointcloud2. However, in R2LIVE and R3LIVE (coming soon), we use the Livox custom message, in which the datatype has recorded the timestamp of each point (used for point cloud motion compensation).

cnhanyuhang, 2021-12-18:
> If the ros message can be used in loam-livox, that means you are recording the pointcloud data in PCL/pointcloud2. However, in R2LIVE and R3LIVE (coming soon), we use the Livox custom message, in which the datatype has recorded the timestamp of each point (used for point cloud motion compensation).

Thanks for your reply, let me record the livox custom msg and try again. Looking forward to r3live!

closed 2022-01-02 by stale[bot]