hku-mars/r2live #72: How to understand formula 23

PengKunPROO, 2022-11-23:
Hi, I'm new to studying slam and its related knowledge, so I'm confused about how to derive the CP_s as below. Why does it need to minus the Ip_C_{k+1} and the Gp_I_{k+1}? Sincerely for your reply, Thanks!
[image]

WFram, 2022-11-23:
It's just point (Ps) projection from global to camera frame. But to make it explicitly dependent on the state vector elements (see IV-C in the R2LIVE paper), corresponding elements (IMU -> Global and Camera -> IMU transformations) should be inverted. That's why rotation matrices should be transposed and translation vectors get negative signs in this derivation.

PengKunPROO, 2022-11-24:
> 

Thank you！ I misunderstand p_C_{k+1} as the corner point's coordinate, so I'm confused about why R_C{k+1} can project the p_C_{k+1} from the camera frame to the IMU frame at the beginning. Thank you for your reply again!

closed 2022-12-10 by stale[bot]